texx00/sandypi #21: GRBL and other controllers compatibility

texx00, 2021-01-04:
The software have been tested only with marlin for the moment.

It should be compatible also with grbl and others but I need someone to test it on real hw (at the moment I cannot try it out on my own)

ubergeekseven, 2021-03-01:
I will test this on my grbl hardware. I might be able to tonight, not sure. It will be soon though. I think there needs to be some merging of projects because there are several that are missing things and together would cover them all.
https://github.com/markroland/sand-table-pattern-maker
This has great designs built in. It does not have an easy method of customizing size. You must manually change variables and I am attempting to make them user accessible in the interface. sandypi could be a front end for playlists for this and this next setup.
https://github.com/jeffeb3/sandify 
everyone must know this one. It could use the code from this project to run the files created by combining the two projects together. Sandify makes it easy to move the created shapes around the table as well as a few other good things. The dev also has a pi image that sandypi should be included in. 

Maybe I can make another post on this in the issues but, figured I would add this considering everything is missing something and this could solve all the missing things by coming together into one package.

texx00, 2021-03-01:
That would be nice! We had some testing done with grbl v1.1, I still need to test out v0.9.

I answered to the other things on the [v1 forum thread](https://forum.v1engineering.com/t/sandypi/19229/29?u=texx)

ubergeekseven, 2021-09-22:
Sorry for not responding to this for so long. I have been using this for my table for  several months and there are some issues I would like to be able to resolve with grbl 1.1. Currently, there is no way to recover from an alarm without manually typing a command. $X is the unlock code and $G is the soft reset command. 
Adding the ability to create macros that are represented as a button would allow for customizing commands for whatever board you run. 
Sometimes, when trying to access the command input box or the menu buttons with the machine running, the terminal scrolls and resets the page to focus on it.

texx00, 2021-09-22:
Thank you for the feedback! Really apreciated!

You think that could be useful to add those buttons in the manual control tab like the "home" or "emergency stop" buttons?
The idea of "macros buttons" is not bad actually. I need to think a little bit how to implement it... Maybe something like 5 "custom buttons", a right click to edit it with possibility to add a custom label and the custom commands? Could it work? 

About the scroll thing: do you have the problem from mobile or from desktop?

ubergeekseven, 2021-09-22:
I only thought of macro buttons because the system could be used with multiple boards using different firmware. If statically set, that would be useless for certain users.
However, maybe when choosing the board used in settings, the macro buttons update to the correct command for each. That is if you think it's easier to implement a static change than something end users could add their own commands.
The scrolling is only on mobile. Usually seeing an issue during the drawing, I grab whatever is closest that's capable of accessing to do the action. On the phone, it always changes the focus to the nearest line of the terminal. However, with the buttons it wouldn't matter anyway.
Thanks for your work on this project. I updated today and see the led control and physical inputs available. Having those abilities are pretty amazing and could expand your platform drastically. If the macros can be triggered by buttons, even better. No device needed at all.

texx00, 2021-09-23:
I think it is easier to have it automatically set by changing the board type in the settings and it is for sure easier to use it like this for beginners... Still I like the idea of the macro buttons and I was already thinking about adding a link between physical buttons and those macro buttons... I need to think a little bit more about this. In this period I'm really busy and will not be able to put them in shortly... I will keep you updated about this ;)

ubergeekseven, 2021-09-23:
@texx00 I would not expect you to have to do anything. I'm glad you've done what you've done. I do look forward to when and if this idea is added. As of now, I can always do what I've done so far.

texx00, 2021-09-23:
Don't worry, I really like the idea... Any idea is welcome, just put them down and we can discuss 😉 this is one of the reason why the project is open: improve thanks to different point of views/experience.

I do have a busy life but with time I will add them in, if they make sense