texx00/sandypi #26: no able to open serial

vincentbraillard, 2021-01-28:
Hello

When start the strat.py I have the following message and I'm not able to connect trough the ip adress:
"
(env) pi@raspberrypi:~/sandypi $ python3 start.py
You are running linux
Preparing start.sh
Starting the server
 * Serving Flask app "server"
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Logger 'app' level: 30 (WARNING)
Logger 'feeder' level: 30 (WARNING)
Serial not available. Will use the fake serial
"
any help?

texx00, 2021-01-28:
Hi,

on the first start it is ok if the serial is not found. After starting the server, you should be able to set which port to use for the serial from the interface: on the top right there is a "settings" button. It will open a page in which you can specify serial port and baudrate to use. Select save and connect to use that serial port. (Probably  I should put this info into the startup instructions... I will leave the issue open until I fix the docs)

The real problem here is that you are not able to load the UI thus you cannot set the port. Can you tell me more about your setup? I can imagine it's a headless raspi... Can you try to connect through _vnc viewer_ or similar and test if locally you are able to load the interface? Are you adding the port to the ip address? If you try with vnc viewer you should open the browser and write localhost:5000. If this one is working it should work also from your computer or phone by putting the raspi ip with the same port like: xxx.xxx.xxx.xxx:5000.

[image]

If you need to see your raspi ip you can use the terminal command "ifconfig".
Let me know if you need additional help.

vincentbraillard, 2021-01-29:
Hello
Thank you very much for your quick answer but I cannot see even the green interface of sandypi. So of course I have try local to connect 127.0.0.1:5000 and also with the raspi ip but both give the same error "ERR_CONNECTION_REFUSED" like there is nothing at this page.

I have followed your instruction and all seem to be ok up to "running server" bellow the code from env start to running serveur:

------------------------------------------------------------------------------------------------------------------------------------------------------------------
pi@raspberrypi:~ $ cd sandypi
pi@raspberrypi:~/sandypi $ virtualenv env
created virtual environment CPython3.7.3.final.0-32 in 1779ms
  creator CPython3Posix(dest=/home/pi/sandypi/env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/pi/.local/share/virtualenv)
    added seed packages: Flask==1.1.2, Flask_Cors==3.0.9, Flask_Migrate==2.5.3, Flask_SQLAlchemy==2.4.4, Flask_SocketIO==4.3.0, Jinja2==2.11.2, Mako==1.1.3, MarkupSafe==1.1.1, Pillow==8.1.0, SQLAlchemy==1.3.22, Werkzeug==1.0.1, alembic==1.4.3, astroid==2.4.2, bidict==0.21.2, click==7.1.2, colorama==0.4.4, future==0.18.2, isort==5.7.0, itsdangerous==1.1.0, lazy_object_proxy==1.4.3, mccabe==0.6.1, packaging==20.8, pip==20.3.3, pip==21.0, pylint==2.6.0, pyparsing==2.4.7, pyserial==3.5, python_dateutil==2.8.1, python_dotenv==0.15.0, python_editor==1.0.4, python_engineio==3.13.0, python_socketio==4.6.0, setuptools==51.1.1, setuptools==51.3.3, six==1.15.0, toml==0.10.2, typed_ast==1.4.2, wheel==0.34.2, wheel==0.36.2, wrapt==1.12.1
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
pi@raspberrypi:~/sandypi $ source env/bin/activate
(env) pi@raspberrypi:~/sandypi $ python3 start.py
You are running linux
Preparing start.sh
Starting the server
 * Serving Flask app "server"
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Logger 'app' level: 30 (WARNING)
Logger 'feeder' level: 30 (WARNING)
Serial not available. Will use the fake serial
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------

and after I try to open localy (I work directly on the Pi4) the adress "127.0.0.1:5000" I have a 404 error and that what I show:
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Not Found
The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------

this is my ifconfig:
pi@raspberrypi:~ $ ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:07:5d:00  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Boucle locale)
        RX packets 36  bytes 3951 (3.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 36  bytes 3951 (3.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.178.56  netmask 255.255.255.0  broadcast 192.168.178.255
        inet6 fe80::e342:faef:e66e:4427  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:07:5d:01  txqueuelen 1000  (Ethernet)
        RX packets 4774  bytes 2705668 (2.5 MiB)
        RX errors 0  dropped 4  overruns 0  frame 0
        TX packets 3434  bytes 550891 (537.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

Can you help me? I really want to test you Sandypi with my sand table... thank you very much

texx00, 2021-01-29:
First suggestion, try to use "sudo python3 start.py". I'm quite sure nothing will change but you nevern know....

If the server is running and you get a 404 error maybe the problem was during the frontend installation.

Try to run again the "installation.sh" script with sudo and post the output here, just to see if there is any kind of error during the installation. Now try again with sudo python3 start.py and check if it works.

If running the installation script with sudo doesn't fix the problem try like this:

From the main folder use the commands:
$> cd frontend
$ frontend> yarn build
Check if you get any error there... If this works without any problem, go back in the main folder with "$ frontend> cd .." and run again the server with "sudo python3 start.py"
If you get an error during the build please copy it here.

In the future I plan to use docker which should fix installation issues like this one... But for the moment this is the only possible way to run it.

vincentbraillard, 2021-01-29:
hello. so I have found some intallation error. I put a copie in a txt file to be readable:

[installation.txt](https://github.com/texx00/sandypi/files/5894917/installation.txt)

texx00, 2021-01-29:
Ok, as I expected is related to the frontend... You are missing npm and consequently yarn

Try to installing by following the instructions of this link or similar: https://www.instructables.com/Install-Nodejs-and-Npm-on-Raspberry-Pi/

After installing this package try to run again the installation script

vincentbraillard, 2021-01-29:
hello
you're the best. it was npm . it was not install.
Now I can see it trough local and network ip!

is it normal that the widget have a turning circle in the midle from a while and cannont see any drawing?

I will open a request topic to have the possibility to read also the .thr file frome sandify!

thank you again

texx00, 2021-01-29:
> is it normal that the widget have a turning circle in the midle from a while and cannont see any drawing?
Can you load here an image?

I know about thr files, I'm already planning to introduce them (#14)... I'm fixing some stuff about playlist at the moment and I'm quite busy. When this topic is done I prioritize the thr compatibility. The only problem I have is that I cannot test it at the moment, my table is corexy...  I will try to do it on the fly but probably I will need some help to test it on a thr table

I will let you know when I finally add it 😉

vincentbraillard, 2021-01-29:
Hi

Yes loading new one work well

Yes I can test it fir you when needed! But I guest led will be  more urgent!

texx00, 2021-01-29:
>But I guest led will be more urgent!

Working on it as well (#23) but the thing there is a little bit trickier... I'm using WS2812B in my build but I think other types might be better... I need to put my hand on test out some WS2801 as well... As you can see there is still a lot to do and my spare time is not that much 😞

vincentbraillard, 2021-01-30:
Ih. Yes I can't believe the amount of work you have to do... Concerning LED, in my case, I just need white led. I don't like RGB and White give better result with better shadows. A simple dimming slide bar would be enough.

texx00, 2021-01-30:
When I get the time I will add a "dimmable" option to the leds branch then... If I remember I will let you know when I do it

texx00, 2021-02-04:
I think this can be closed now that the readme is updated

closed 2021-02-04 by texx00