texx00/sandypi #27: no able to use with GRBL 1.1

vincentbraillard, 2021-01-29:
Hello again

I wanna to test with my board GRBL and it seem to no work. Following error is showing:

N2 M110 N2 *93
N3 M220 *0
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]
error:9
> $x
N5 $X *7
okokokokokerror:9
error:9
> G1 X1.5
N7 G1 X1.5 *93
error:9
error:9

texx00, 2021-01-29:
I do not have experience with grbl but I think I will need to flash it on something to test it better with sandypi... (#21)

I searched online and I can see error:9 may be related to [homing problems](https://community.carbide3d.com/t/homing-error-grbl-error-9/9881/7)... Are you sure the board is working correctly? If you didn't already try to use another software like [repetier host](https://www.repetier.com/). This way I can be sure the problem is within sandypi and not about your build settings

vincentbraillard, 2021-01-30:
Hello. No my GRBL works perfectly. I use it with cncjs already. I don't know what don't work but normaly, after the connection, Grbl send to the uart all $ code and his value! and that I cannot show with sandypi

texx00, 2021-01-30:
Ok, I will have a look when I can. Thank you for reporting 😃

texx00, 2021-01-31:
I've got some time to check it out.
I installed plain grbl 1.1 on an Arduino UNO i had around but I couldn't try the real hw, only the communication with the board.
I documented myself about grbl a little bit more and I can see it uses special commands that starts with "$".
The problem was that the software was adding always the line number, also for the $ commands.

I created a new branch with a fix. If you can test it for me, I will merge it to the master when I'm sure it's working.

To run the new code version from the branch:
```
(env) $> git pull
(env) $> git checkout grbl_support
(env) $> git pull
(env) $> sudo sh install.sh
(env) $> sudo python3 start.py
```

Still the software is not setup for thr files so I think you won't be able to use sandypi yet.
I might have some spare time this week. I will see if I can at least start to have a look at them... I think it will not require enourmous changes...

vincentbraillard, 2021-01-31:
Hu. You're a master...If I'm able to put the new branch with commande line
(I'm not a master in commande line) I will test it tonight!

No I you GCODE file. But because I'm using a scara robot, I need to export
thr file from sandify and run a small python software to make a .gcode file
compatible for scara. So yes I will be able to work with sandypi. Just the
small widget will not work but that is not a priority....

Will keep you in touche if it’s work😀

vincentbraillard, 2021-01-31:
Hello
So I have follow your instruction but now I cannot any more connect to sandypi trough 127.0.0.1:5000 or local ip adress. Yesterday the connection was ok.

Bellow is what I have done:

[grbl_installation.txt](https://github.com/texx00/sandypi/files/5900336/grbl_installation.txt)

I think the installation was ok but may be you can see something. I have done it 2 time to be sure!

texx00, 2021-01-31:
So, I had a look at the log and it is quite strange....
When you run the server it says a library is missing but that one was installed succesfully in the steps before....

Try to run `$> sudo apt install python3-netifaces` and then run the program (no need to run install.sh this time, just use `sudo python3 start.py`

If this is not working try with: 
```
(env) $> sudo pip uninstall netifaces
(env) $> sudo pip install netifaces
(env) $> sudo python3 start.py
```

Let me know if one of the two works...

Additional detail: if the server starts, the interface may not load and it will get stuck on a gray page... In this case clear the browser cache and data or follow the instruction in the "Update" section of the readme

vincentbraillard, 2021-01-31:
Hello. So because your method didn't get any result I decide to reinstall all the Raspberry. No I can connect again succefully to the GUI.

I have connect the device suceffully like you can see bellow:
[image]
but when I try to send command it don't work:
[image]

I can do some more test if you want

texx00, 2021-01-31:
Yes, the sw sends some periodics command to check the status of the device (only solution I found to fix a problem I found some time ago)...

I think the first problem here is the checksum: in the first picture (settings page) on the left of "save and connect" there is a switch: turn it to be grey to disable the chechsum (first problem should be fixed: you should not get error:20 anymore)

vincentbraillard, 2021-01-31:
So

First I disable the checksum but I have still error 20 that come sometimes

second. I Have test without the $ but nothing happen. 
"
Grbl 1.1h ['$' for help]
Grbl 1.1h [error:20error:20
> X
N5 X
error:2
> G1 X1
N6 G1 X1
error:22
> G28
N7 G28
> G92
N8 G92
error:26
error:20
"

I have deseable the homing in grbl so this is not the issu. I use cncjs and the onlything I need to do is $X to put the state to Idle!

When I connect trough cncjs, the first answer from grbl after the connection is that:

vincentbraillard, 2021-01-31:
so puting F20 in the box "command before drawing" All start good. but after few time "40sec" it stop and in the manual command line I see it's because error 20 is showing!

this Gcode works well so it don't come from there!

The strange thing is the drawing restart by itself after few sec/min....

the checksum is desable!

this is the Gcode:

[start0.8V2.Gcode.txt](https://github.com/texx00/sandypi/files/5900877/start0.8V2.Gcode.txt)

and this is a copie of the command line from sandypi

[Gcode from manual command.txt](https://github.com/texx00/sandypi/files/5900886/Gcode.from.manual.command.txt)

hope it help you

vincentbraillard, 2021-01-31:
Here is a guide line for grbl
https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl
You may be should put a single switch in setting to choose between marlin abd grbl. So it would be easier for you and all command would wirk like the stop or emergency stop!

texx00, 2021-02-01:
Thank you for the testing! 😄 

> You may be should put a single switch in setting to choose between marlin abd grbl. So it would be easier for you and all command would wirk like the stop or emergency stop!

I think it makes sense... I wanted to keep separate options because the system may be more flexible. On the other hand, it is not only the checksum as you sad (emergency stop, etc...) so to make it easier for the user I will put a select menu from which it is possible to choose the firmware and that will setup everything (I will prepare Marlin and grbl for the moment, in the future it will be easy to add more)

I'm planning also to put a select to choose between cartesian, polar, scara and other options... This will require some work to create a correct drawing preview both for the uploads and the manual control... Not something I will be able to do in only few days...

> this is the Gcode:
start0.8V2.Gcode.txt

I had a look at the file and I think the last line may have something to do with the machine stopping... It includes a comment in the command but my software is not setup to recognize it (it checks only full line commands, not in line commands)... I will add a check to remove possible inline commands before sending the gcode

> when srial connected, checksum desable, every 30 seconde precisely I have a error20 showing on the serial

I think I know already where is the problem... I will fix it as soon as possible and let you know to try out if it fixes the other problems (drawing reliability and start of next drawing of the queue)

> stop drawing don't work
Emergency don't work

I put them there in the first version of the frontend but now that it is written in react I think I forgot to test them even on Marlin 😬 I will try to put it to work together with the select button fix...

Let me some time to arrange the fixes, I will let you know when and how to test the changes 😉

vincentbraillard, 2021-02-01:
Hi. Thanks for your reply. For your information, even after erasing the last line, at the end if the drawing, sandypi keep in stat now drawing and stop drawing doesn't work!

Thank again for your work

texx00, 2021-02-02:
Ok, I think I fixed everything in the list (up to this moment at least)

When you have the time, can you try the update for me? I'm still using a hardware-less board so I don't know really if it is working correctly 100%.

I fixed the stop issues and now it is working correctly also with the playlists.

> During drawing, sometime the process stop 20% and don't restart, sometime the process go to the end 80%. I don't have found a way to found what's happen when it stop. Is there a way to check the serial?

I hope this one fixed by itself with the other stuff because otherwise I don't know what I can do without real HW. In the future I will for sure put together something (maybe a scara or polar build?)

To run the new update:

```
(env) $> git checkout grbl_support (you should be here already)
(env) $> git pull
(env) $> sudo sh install.sh
(env) $> sudo python3 start.py
```

If it is running but it doesn't show up in the browser clear the site cached data or run `localStorage.clear()` in the browser javascript console (I will fix this in an upcoming commit I hope)

vincentbraillard, 2021-02-02:
Hello

Thank you for all the work. Bit I have a problem to test:
I have update the grbl_support and it's not anymore possible to connect to the ip adress

So I have reinstall all the raspberry from the standard image, and install you installation following the master branch. Connection was ok. but after maked the installation again of the grbl-support, the wer server doesen't work again....

Have you touch something here? bellow the code when I try to start the server:.
 
pi@raspberrypi:~/sandypi $ sudo python3 start.py
You are running linux
Preparing start.sh
Starting the server
 * Serving Flask app "server"
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off

To run the server use 'ip:5000' in your browser with one of the following ip adresses: ['127.0.0.1', '192.168.178.102']

Logger 'app' level: 30 (WARNING)
Usage: flask run [OPTIONS]

Error: While importing "server", an ImportError was raised:

Traceback (most recent call last):
  File "/home/pi/sandypi/env/lib/python3.7/site-packages/flask/cli.py", line 240, in locate_app
    __import__(module_name)
  File "/home/pi/sandypi/server/__init__.py", line 63, in <module>
    from server.hw_controller.feeder import Feeder
  File "/home/pi/sandypi/server/hw_controller/feeder.py", line 14, in <module>
    import server.hw_controller.firmware_defaults as firmware
  File "/home/pi/sandypi/server/hw_controller/firmware_defaults.py", line 1, in <module>
    from dotmap import DotMap
ModuleNotFoundError: No module named 'dotmap'

texx00, 2021-02-03:
> Thank you for all the work. Bit I have a problem to test:
I have update the grbl_support and it's not anymore possible to connect to the ip adress

Can you explain better what you do and what happened? With picture also if it happen again if you can

To fix the error above use `(env)> sudo pip3 install dotmap` or `(env)> sudo python3 -m pip install dotmap.` I forgot tu put it in the requirements.txt used in the installation probably... After that it should work... I think the error was the same when you tried to run the update the first time

vincentbraillard, 2021-02-03:
HI
So I test install manualy dotmap but no success to strat the server... I have made a screenshot video so you can check what happen. it's a little bit long but you can scroll the video...

https://www.youtube.com/watch?v=hU4Mf-n0wJ0

texx00, 2021-02-03:
Really strange!

I will update the requirements file and let you know to try again (that way
it must work)

One last try if you want is to run "pip3 install dotmap" without the sudo
word (or python3 -m pip install dotmap)

You can also wait for the update... I will do it this evening when I get
back home

vincentbraillard, 2021-02-03:
Hi. don't take to much time for that, I make it work now....
I'm currently testing all features to give you a feed back this evening...

texx00, 2021-02-03:
I was on my phone when I answered and I couldn't see very well the video...

No I had a look at the commit and the file is correct actually. I think the problem is in the "sh install.sh" command. I need to update the readme file to use "sudo sh install.sh" instead I think.

> UPDATE:
WORK NOW. testing now

How did you manage to fix it?
Is Grbl working better now?

vincentbraillard, 2021-02-03:
Hi. So:
I had issue with some stop during some drawing. Or a drawing go fine until the end, or the a drawing stop many many time an restart by itself. I suspect a 1.5m usb cable and now I have change it but I'm not sur....I have made a video to shiw you what happen. May be you can see something wrong!
https://youtu.be/-H7x3QL40qE

-Stop drawing: work well now even if we need to refresh the page to clear the "now drawing"
-playlist: work but to do a playlist you need to add draw from drawing tab. From playlist tab, the add drawing doesn't work
- installation. I have delete the whole folder sandyfi and reinstall all with sudo. 
- Name of drawing. Could you just add name of drawing in playlist and "now drawing' tab. Because I use transformed gcode for Scara "thank Jeffeb3 for the small phyton code", I can't see the drawing and text would be helpful (in my case)

DO YOU WANT SOME MORE TEST?

- IDEA TO IMPROVE! 
 1) add a slider to modify the speed/feedrate during drawing (F10 to 70) in Grbl case
2) a way to pause the drawing and not only stop it

Thank again and tell me if you want some particular test. I will continue test with new short usb cable!

The following draw waa made with sandypi and scara

[image]

texx00, 2021-02-03:
Thank you man! You are doing a great job! 😄 

From what I can see in the video I think it is really a comunication issue. Not sure if it could be the cable or if the serial communication is not managed correctly by the software. 
The problem here is that sandypi waits for an "ok" message before sending new messages (grbl uses a 16 positions buffer, if it fills it will give an error or skip commands). Sandypi is waiting for those "ok" to clear the buffer but if it receives "okok" or "k" because of communication issues it will not count the message for clearing the buffer.

The drawing stops because sandipy is waiting for the buffer to be cleared but it doesn't receive all the acks and the messages accumulates.
I introduced I timeout that check the buffer status (status report in grbl) to check how many lines are in the buffer. The timeout is set to 30 seconds at the moment. I can reduce it to be something like 1-2 seconds and it may make the drawing smoother, without that many stops.

My build is much more reliable under the communication point of view (don't know why really). It happens sometimes that one message is lost and the buffer get stuck but it happens rarely. On the other hand, on Marlin I cannot ask for a status report (or better, I looked for it but I didn't find a way of doing it) so the only thing I can do (at least at the moment) is to manage the buffer with the acks that means I need to keep a longer timeout to avoid having errors of full buffer.

For sure I need to experiment more with the part of the software that feeds the gcode but it is not really easy and it requires to have the hw on hand (which can be difficult with my job, I travel a lot). I will add a check to the firmware type and reduce the timeout value for grbl (for the moment at least).

For the moment I updated the timeout in the last commit. Should be enough to run: 
```
(env)$> git pull
(env)$> sudo sh install.sh  (not really necessary this time, but just to be sure)
(env)$> sudo python3 start.py
```

If after this one the board is drawing without stopping I will close the pull request and the issue (to be consistent with the issue title)

I updated the instruction in the readme file (only in the branch at the moment) to use the sudo command on the installation part. I looked around and it is really related to how sudo works and there are some workarounds but they require some testing and at the moment I prefer to focus on making the sw usable (at least for you with grbl 😉 )

> -playlist: work but to do a playlist you need to add draw from drawing tab. From playlist tab, the add drawing doesn't work

I will have a look

> Name of drawing. Could you just add name of drawing in playlist and "now drawing' tab. Because I use transformed gcode for Scara "thank Jeffeb3 for the small phyton code", I can't see the drawing and text would be helpful (in my case)

I'm going to ask him if I can use it in sandypi to load the files directly there. I will add the name temporarily.
I will also fix the visualization by adding a "cartesian, polar, scara" selector to choose for the table type so that the preview can be drawn correctly but it will take some days of work with the little spare time I have.

> add a slider to modify the speed/feedrate during drawing (F10 to 70) in Grbl case

Already on the list 😉 

> a way to pause the drawing and not only stop it

I will put a button in the queue page

> The following draw has been done with sandypi and scara

NICE! 🔝

vincentbraillard, 2021-02-03:
Thank man...

If you want to see the python code to make a scara from thr file so you may be use it to make a reverse to draw!

[thrToGcodeforScara.py.txt](https://github.com/texx00/sandypi/files/5921925/thrToGcodeforScara.py.txt)

Will make some more test following days concerning serial issues!

texx00, 2021-02-04:
I changed my mind already... The fix I did yesterday it's going only to
shorten the time the board is stopped... I will fix it better this evening.

About converting thr files I see jeff is adding it directly to sandify... I
will just include the preview selection, non the conversion...

https://forum.v1engineering.com/t/polarsandtisch/20102/61?u=texx

vincentbraillard, 2021-02-04:
Hi. so even better ;-)

I don't know how to pull new request so I Write it here! I forgot to tell you yesterday some interesting feature to add!

- When playing playlist, shuffle mode and repeat mode would be great
- To be able to make some pause between drawing. I imagine a slider from 0 (no pause) to X minutes. So we can enjoy our beautiful draw
- The possibility to add some physical button (play, pause, light on/off) so we can without smartphone directly pilot the table. Actually my wife is asking it ;-)

Thank, I will test yours update this night!

texx00, 2021-02-04:
So:

I pushed some changes. I think i fixed the playlist issue (it was loading the drawing but not refreshing the playlist) and improved the timeout (now even if it stops it will stop for a brief moment only (it should not stop anymore really I think)

To run the update the usual:
```
(env) $> git pull
(env) $> sudo sh install.sh
(env) $> sudo python3 start.py
```

About the other message:

> When playing playlist, shuffle mode and repeat mode would be great
To be able to make some pause between drawing. I imagine a slider from 0 (no pause) to X minutes. So we can enjoy our beautiful draw

I was working on this before you opened the issues... Indeed the custom command you can add into a playlist is the first step forward more options.
In my list I have: pause a playlist for a certain time or until an hour (so that you can wake up with a fresh drawing without drawing the entire night for nothing), start another playlist, start a random drawing from the playlist or from the full list, move along the borders between different drawings, change leds animation/color between drawing...
For sure the list will get longer once other people start to use sandypi...

> The possibility to add some physical button (play, pause, light on/off) so we can without smartphone directly pilot the table. Actually my wife is asking it ;-)

Quite easy to do actually but it requires some time to make it work (read as debugging time)... Though, I think there are more important features that I need to focus on for the moment. If you like the idea you can open a new issue for this. I will mark it with a "first issue" label so that if somebody else want to help in developing the software can start with this one maybe...

> Adding a 5way led strip (RGB +WW + CW)

Maybe you notice already: in the latest versions I made you install there is a LED tab... I was working on this one as well before focusing on grbl... If I have time during the weekend I will try to fix the dimmable option (it's not working for sure, it's just a placeholder for the moment) so that you can at least control the white led strip you asked before (only 1 pin control for the moment). In the future It will be possible to add also RGB  and the extra pin control but, again, I want to focus on something more important before...

Still, if you have ideas don't worry and write them down... Better to list them twice than never 😉 

Let me know if this version is better... In my hw-less build it looks like it's not stopping but it's not reliable this much...

texx00, 2021-02-06:
I have news! I finally managed to add a preview for scara robots...

I did this in another branch because I didn't know if the grbl branch was good enough...
It is in another pull request (#29)

You can test the new changes with:
```
(env) $> git pull
(env) $> git checkout drawing_preview
(env) $> git pull
(env) $> sudo sh install.sh
(env) $> sudo python3 start.py
```

I added some additional settings that needs to be setup correctly to work with the scara... I put the values from the conversion script as a default. I will add some desciription and a wiki about how to set those settings in the coming days, but first I need to upgrade the settings page in the background and fix the pagination which sucks at the moment.

Let me know if it works! I did also some small changes to the connection with the board... Let me know if it works better or not... (when you have some time of course, no hurry)

vincentbraillard, 2021-02-07:
Hello

Sorry for the small delay. I have done a lot of test to try to understand why it's doesn't work.

For the GRBL serial issues. SO the last update Was no very better. When the drawing stop, it is impossible to restart, May be it was better when it restart by itself like before.
Sometime, GRBL ready a wrong value putting the scara arm away from the trajectory. it's like the coordinate are G1 X2 Y2 and Grbl Read G1 X2 Y200. It happen 3 times
When the drawing stop, it's not possible to write manually a command because it need to clear the buffer before. it because it don't move, it never clear the buffer.
It's a problem very difficult because it's very random. I can start 2 times the same drawing, one will be perfect, the other don't.  I'm pretty sure it's a serial communication issue. Even with a new short USB cable.
Is there a way to check the code is it transmit to GRBL. May be we could find what doesn't parse right! 

On thing stange may be can give you a idea is when connecting, GRBL sand always the GRBL version follow with all parameter value. And this I don't see when I connect.

On the manual control windows, I still see every 10 sec around an "error 24 - More than one g-code command that requires axis words found in block."

Will try now the drawing

texx00, 2021-02-07:
I have been working on the communication today...
Yesterday I saw the errors in the log also on my board but I was hoping they were not that bad...

Today I reworked quite a lot the part about the connection: it was a little messed up but with marlin it was working fine. Now it is much more under control and in my case with grbl is giving only few errors now and then but I think the problem there is really comunication issues due to the communication and I cannot improve them more than this unless I build a piece of hw to test it out better. Nonetheless, with this update the result should be much better! (I hope at least). I fixed also the stop button.

The problem with grbl is that it doesn't have a checksum like marlin does. For this reason it is much more difficult to handle communication errors... If it happens again that the controller doesn't get a value correctly it will still use it moving the axis in the wrong position. Anyway, this update should be much better I hope... I had some troubles with small geometries before also on marlin but with this version it's much more stable/smoother/faster. Still the scara configuration uses a lot of commands in a short time so I think it may be even more sensitive to this problem so let me know...

I published the update to the previous branch:
```
(env) $> git pull
(env) $> git checkout grbl_support
(env) $> git pull
(env) $> sudo sh install.sh
(env) $> sudo python3 start.py
```

In this update I fixed also some other bugs: I hope the installation is going to be successful (I tested it on my raspberry so it should work fine)

> Sorry for the small delay.

Don't worry! Do this only when you have the time, it's not urgent, there is no life in danger 😉

vincentbraillard, 2021-02-07:
Hello

So I install and test the drawing_preview branch. And GRBL communication work way better. I test it this afternoon. I don't know will installing the drawing_preview branch it modify also the GRBL communication but it work way better.

Sometime it cut during less than 1sec and restart.

Concerning Drawing:
It's work very well. congratulation! I just needed to reboot the PI to make it worked well, I just notice few thing:
1) the "now drawing" keep even when the drawing is finish. I need to reload the page.
2) When you modify by clickAndDrop the order of the playing, you need also to reload the page to take in. Only save playlist isn't enough
3) May be a color contour around the current drawing in the playlist could be helpful to see witch is drawing
4) From playlist, there is no way to import gcode from drawing tab. you need to dowload it. but maybe it's no yet done ;-)

I thing you can merge to master all improvement because it work well now! And so you can keep going you todo list!
I will now install endstop for homing on my scara and test the home button!

Tank you again

vincentbraillard, 2021-02-07:
[image]

texx00, 2021-02-07:
Perfect!
Then I will merge later today probably... Thank you for the help! 😄 

If you find more bugs please let me know...

> May be a color contour around the current drawing in the playlist could be helpful to see witch is drawing

Good idea! I need to change few things to make it work though...

> From playlist, there is no way to import gcode from drawing tab. you need to dowload it. but maybe it's no yet done ;-)

You are right, at the moment it is possible to do this from the drawing list but not from the playlist directly... I will think about it.

> * the "now drawing" keep even when the drawing is finish. I need to reload the page.
> * When you modify by clickAndDrop the order of the playing, you need also to reload the page to take in. Only save playlist isn't enough

I will have a look

closed 2021-02-07 by texx00

vincentbraillard, 2021-02-14:
UPDATE

I Have restart the Raspberry and GRBL doesn't work anymore. Don't now if we need to restart it at each update butnow during a GRBL drawing, it stop. in manual windows there is no error and in the command windows I can see that when it stop!

ERROR:werkzeug:Error on request:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 323, in run_wsgi
    execute(self.server.app)
  File "/usr/local/lib/python3.7/dist-packages/werkzeug/serving.py", line 312, in execute
    application_iter = app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/flask/app.py", line 2464, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/flask_socketio/__init__.py", line 46, in __call__
    start_response)
  File "/usr/local/lib/python3.7/dist-packages/engineio/middleware.py", line 60, in __call__
    return self.engineio_app.handle_request(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/socketio/server.py", line 560, in handle_request
    return self.eio.handle_request(environ, start_response)
  File "/usr/local/lib/python3.7/dist-packages/engineio/server.py", line 374, in handle_request
    socket = self._get_socket(sid)
  File "/usr/local/lib/python3.7/dist-packages/engineio/server.py", line 562, in _get_socket
    raise KeyError('Session is disconnected')
KeyError: 'Session is disconnected'
WARNING:engineio.server:Invalid session 38ba82e7151e4cdaa7e008cd6bb3a17e
WARNING:engineio.server:Invalid session 38ba82e7151e4cdaa7e008cd6bb3a17e

PLEASE STILL DON'T MERGE

PR #28 (2021-02-07): Adding Grbl compatibility
Grbl doesn't like the checksum that is necessary in Marlin.
Added a new checkbox to the settings to enable/disable the checksum
Also, grbl requires special commands starting with `$` without line number. Now the software is filtering out those commands to avoid troubles

Didn't tested it on real hw for the moment. Need some feedback

Will fix #27 
Related to #21