texx00/sandypi #33: Trouble starting server

Louhacking, 2021-02-22:
Hello 

I've been trying to get the server to work, however I'm not able to get it to work. I've read the other issues and the troubleshoot page but it still not working. I get this error message where it is talking about not able to find the module named GPIO. Can you help me?

[Troubleshooting.txt](https://github.com/texx00/sandypi/files/6022213/Troubleshooting.txt)

Thanks in advance, Lou

texx00, 2021-02-22:
Hi, the GPIO module is something I need to include for the leds control. I started it but I switched to other works so the error should be there but should not cause anything (should show the error but keep on going with other stuff).
2 questions to help me out:
 * When you run the script does it stop on its own or will it continue to run without additional messages?
 * Can you check if npm is installed correctly with `npm -v` in the command line?

If this doesn't work, try to change the serial port setting manually in the "server/saves/saved_settings.json" file. You can list the available serial port with a bash command but I do not recall by mind now and I'm away from my computer... Put the correct value (something like "tty/usb0") in the place of "FAKE"

Please keep me updated 😉

Louhacking, 2021-02-22:
hi,

Thanks for the quick reply!

When I run the script, it gives me the the errors like I showed in the text file above and I can type new commands ( I'm a really new to this ) When I check the if npm is installed it returns: 6.7.0. 

I've tried to change the serial port within the saved_setings.json file, however I'm not able to get it to work as I don't really know how. The correct value should indeed be: /dev/ttyUSB0. 

Thanks in advance, Lou

texx00, 2021-02-22:
Hi again,

I just uploaded a small update to remove that package from the imports. You can try it out with (from within the virtual environment on the master branch):
```
(env) $> git pull
(env) $> sudo sh install.sh
(env) $> sudo python3 start.py
```

When you do this save the output of the installation in a text file. If this update doesn't fix the problem send me the installation log + the new log when you start the server. This time you should get only one error (I removed the code that was causing the GPIO error) about the serial connection but the server should keep on running and you should be able to reach the interface at the give ip address.

Let me know if this update fix the problem or if you get something else...

Louhacking, 2021-02-22:
Hello,

I still can't access the interface, but the GPIO error is gone, these are my logs:

[installation.txt](https://github.com/texx00/sandypi/files/6025717/installation.txt)

[serverstart.txt](https://github.com/texx00/sandypi/files/6025718/serverstart.txt)

texx00, 2021-02-23:
I found the problem: during the installation there is an error in the build of the frontend.
```

----- Installing js dependencies and building frontend app -----

> yarn@1.22.10 preinstall /usr/local/lib/node_modules/yarn
> :; (node ./preinstall.js > /dev/null 2>&1 || true)

/usr/local/bin/yarn -> /usr/local/lib/node_modules/yarn/bin/yarn.js
/usr/local/bin/yarnpkg -> /usr/local/lib/node_modules/yarn/bin/yarn.js
+ yarn@1.22.10
updated 1 package in 35.767s
yarn install v1.22.10
[1/4] Resolving packages...
[2/4] Fetching packages...
error An unexpected error occurred: "/usr/local/share/.cache/yarn/v6/npm-@babel-runtime-corejs3-7.12.13-53d09813b7c20d616caf258e9325550ff701c039-integrity/node_modules/@babel/runtime-corejs3/.yarn-metadata.json: Unexpected end of JSON input".
info If you think this is a bug, please open a bug report with the information provided in "/home/pi/sandypi/frontend/yarn-error.log".
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
yarn run v1.22.10
```

To fix this you can try like this:
from within the sandypi folder:
```
$> cd frontend
$> yarn cache clean --all
$> yarn install
$> yarn build
```

If it gives you the same error try it once again. 
If doesn't work still you can try also to run: 
```
$> npm cache clean --force"
$> npm install -g yarn
$> yarn install
$> yarn build
```
If you keep getting an error post the result of the commands as well as the error log file in "in "/home/pi/sandypi/frontend/yarn-error.log".

If the commands above ends correctly, it will be enough to go back to the main folder and start the server:
```
$> cd ..
$> source env/bin/activate
(env) $> sudo python3 start.py
```

Louhacking, 2021-02-26:
hello again, 

I'm still getting a error when i tried the comand: build yarn, the error is this:

pi@raspberrypi:~/sandypi/frontend $ yarn build
yarn run v1.22.10
$ react-scripts build
Creating an optimized production build...
The build failed because the process exited too early. This probably means the system ran out of memory or someone called `kill -9` on the process.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

could the reason be that i'm using the raspberry pi zero w?

texx00, 2021-02-26:
Yes, I think so... The build process can be heavy for the pi3 already. If
you want, you can try to build that part on a computer running windows or
linux and then you can copy the build directory in the same place into the
pi (never tried this but theoretically should work). I will have a look if
it is possible to make the build process lighter to make it compatible also
with the zero but for the moment it is not supported directly...
If you want to try this way on the zero and you cannot figure it out on
your own let me know... I will try to give you more detailed info...

texx00, 2021-02-26:
I investigate a little bit more and it may be possible to make it work on the zero as well but I will need to test it out by myself.
I will not have the time to fix it in the short term. The problem is with the `yarn build` command which is using too much memory. There are some flags fro the build-script that may fix the issue. 

I changed the name of the issue. If somebody would like to have a try to fix the problem I will accept it.

If the flags do not solve it may be possible to create a guide in order to build the sw on a computer and then copy the necessary files on the raspi. This will require a custom installation script in order to skip the frontend build process on the zero with at least a message that tells to have a look at the right procedure.

Louhacking, 2021-02-27:
Thanks for your time! I really appreciate it. Lets hope we find a solution soon.

texx00, 2021-02-28:
So, I have news...

I had the time today to play with my zero W and I think the build process cannot be done on that board in any way (not enough power).
There is big but: if you have a pi3/4 laying around, you can take out the sd from the zero, put it in the other, run the installation there and than put the card back into the zero. I tried like this and it is working...

In the future I think the best solution will be to release an image or create a branch with the frontend files already built (in this way the installation will be quicker also).

If you cannot run it on a pi3/4, now I'm going to add some more details in the readme file on how to build it on a computer and the add only the necessary files.

Louhacking, 2021-03-01:
Luckily my housemate had a RPi2 laying somewhere around, so I was able to compile the build. So it works for me for now, again thanks for the time!

texx00, 2021-03-02:
Perfect! I think it will be slow in doing certain things (like uploadong a drawing)... Let me know how it works and if you have issues

Louhacking, 2021-03-02:
Hi, I need to use baudrate 250000 for my ramps 1.4 board, however it's not available within the settings tab.

texx00, 2021-03-02:
Hi! 250000 is not in the default baudrates around the world. Still, with a quick search I saw that marlin is using it to reduce the error rate in the communication (I think it is related to the 16KHz clock but not completely sure). I will add it in the list of possible baudrates even if it is not considered to be a standard one because other peoples may be using it. 
Until I put it in, you can add it in the "server/saves/default_settings.json" and in the "server/saves/saved_settings.json" files.
You will have something like this:

```
        "baud": {
            "name": "serial.baud",
            "type": "select",
            "value": "115200",
            "label": "Serial baudrate",
            "available_values": ["2400", "4800", "9600", "19200", "38400", "57600", "115200", "230400", "460800", "921600"],
            "tip": "Select the correct serial baudrate"
        },
```

In the available values, after 230400 add the correct value between brackets (be careful with comas as well). Then, restart the python program, refresh the web page and the option should appear.

Let me know if the trick works and also if the communication is working even with a non standard baudrate

Louhacking, 2021-03-03:
The trick works! My table is now fully working.

texx00, 2021-03-03:
Perfect!
Since I did include the guide in the readme I'm going to close this issue.

commit 62f3cdcb (2021-03-03): Added 250000 baudrate for marlin. Closes #33
closed 2021-03-03 by texx00