texx00/sandypi #34: new playlist with shuffle mode, repeat, timing ...

vincentbraillard, 2021-03-01:
Hello

I'm testing now the new update of the playlist a saw following issue/improvement:
1) When you want to add a new drawing from the playlist, you can only update from local folder. It would be much better to add new drawing from already imported drawing from drawing tab!
2)adding custum Gcode. ok could be helpfull for someone
3) when adding timing element, a windows open to know the palylist as change and want to know if you want to save or not. But before adding the elements I have save the playlist so there is a problem
4) timing type are quite well! good job
5) random: I don't understand how it work. When I add a random, it add me to the queue the same playlist and that's all. I see also sometime you call random and sometime shuffle
6) Start new playlist. I also don't understand very well what is it. It's a kind of repeat mode? 
9) You cannot add a new element between to drawing. this could be helpfull
7) IN GENRAL: 
Could you apply a save eachtime you do something so you don't need to press save playlist or have a popup windows
-To be sincery with you, it's quite hard to use. it would be, in my mind, much better to put two icon, one for shuffle and one for repeat in the "now drawing" windows. just like all music player application. if you cannot add to the "nw drawing" windows so you can add in the home of playlist tab to the right of the playlist name two icon for repeat and shuffle.
- for pause. Like I say it would be more easy to add a slider(in setting or playlist tab or now drawing) from 0 to x minutes who apply a pause between all drawing that adding eachtime a timing element.

I think the base is well. but the way to use it is not easy. What do you think about my proposition?

texx00, 2021-03-01:
Hi! Thank you for the testing!

> 1. When you want to add a new drawing from the playlist, you can only update from local folder. It would be much better to add new drawing from already imported drawing from drawing tab!
2)adding custum Gcode. ok could be helpfull for someone

I will add it, I need to think how to implement it

> 2. when adding timing element, a windows open to know the palylist as change and want to know if you want to save or not. But before adding the elements I have save the playlist so there is a problem

I will have a look

> 3. timing type are quite well! good job

Thanks, almost surprised something works straight away 😮 

> 4. random: I don't understand how it work. When I add a random, it add me to the queue the same playlist and that's all. I see also sometime you call random and sometime shuffle

I will check the names and desciptions to make it easier to understand. The idea is this:
two possibilities:
 1. select a random drawing from the full list of available drawings
 2. select a random drawing between the one in the current playlist.
The idea is you may not need to play a shuffle of all the drawings but only for the single playlist (like you create a christmas playlist and you want to shuffle between those only).

> 5. Start new playlist. I also don't understand very well what is it. It's a kind of repeat mode? 

Yes, it can be used as a repeat mode or it can be used to start another playlist. The thing about this element is that I have some additional options to add in the future.
Still, with this element it is possible to create a continuos drawing schema by adding the following elements to the playlist:
 * random drawing element
 * timing element with a delay
 * start playlist (the same playlist)

Then if you add drawings after the start playlist you can put a sort of "topic" and the table will draw only from those (if within the random drawing element you select the playlist as target instead of all the loaded drawings)

You can see this result in a loop running continuosly new drawings. This will be more or less the same result that will be seen in the queue when I add the the "start random" or "shuffle drawing" button in the main page.

Still your point means they are not intuitive to use. I will think about better descriptions for sure, and also it may be is better if I create a guide on how to use them as well.

> 7. You cannot add a new element between to drawing. this could be helpfull

The elements can be sorted with drag and drop, thus even if an element is added at the end of the playlist it can be moved in the wanted position

> 8. Could you apply a save eachtime you do something so you don't need to press save playlist or have a popup windows

This is something I'm very conflicted. I didn't do it yet because I was thinking "what if I just want to go back and undo the previous changes?". This is why I was not saving after every change. Still, as you said, it is not confortable to save every time. Maybe it is better if I save automatically every change and remove the possibility to undo them (it will not be that easy to implement an undo "step-by-step" process after saving with the current code). I will put it in...

> To be sincery with you, it's quite hard to use. it would be, in my mind, much better to put two icon, one for shuffle and one for repeat in the "now drawing" windows. just like all music player [...]

You are totally right, but this is for the next step. The elements design was intentional to have something with a lot of costumization at the back, then the easy buttons will be added to create certain types of elements automatically. The buttons you are mentioning will be a shortcut to create those elements that are required to manage the queue of drawings. The result is: when I will add the "shuffle play" button (very soon I hope) the software will add those kind of elements in the queue automatically without creating a custom playlist. Still, if you want it is possible to use them in the playlists... This is a "playlist upgrade" for this reason. 

I'm still thinking about the playlists preview also: as you shown them in the picture it can be intersting but I'm not really sure if I would put multiple buttons in a preview. Still, those button can go inside the playlist and in the preview I can add a right click to "start" or "start shuffle". For the moment I think I will do it like this, then we will see and discuss the usability of this choice.

Let me know what you think about my comments 😉

vincentbraillard, 2021-03-02:
Hi. Your welcome. It's the min I can do for all your job!

It was a point in more you didn't see in my last comment:

7.during a run of the playlist, when pressing the stop button it begin the nextone, so I think you could change the stop for a "next" and add a new button who clear queue and stop!

ELSE
ok if for you it's a step letting full possibility it's nice. At the end (in my mind only) you could keep custom gcode ,start new playlist and add new drawing in the add option windows. Else (shuffle and repeat) would be nice to be set in now drawing!

Why in now drawing? Because like when you hear a music playlist you can decide during the run of the playlist if you want to repeat it or shuffle the queue drawing....

For me this would be the perfect combination!

But we can first test a lot like that and see if there is any bug😉

vincentbraillard, 2021-03-02:
And concerning the save issue without undo is not a problem.. the worth thing can happen is the customer need to do again the playlist...so no a big problem 😉

texx00, 2021-03-02:
Sorry about the missing point...

> 7.during a run of the playlist, when pressing the stop button it begin the nextone, so I think you could change the stop for a "next" and add a new button who clear queue and stop!

Yes, you are totally right.... I was thinking the same lately... Actually the entire queue page is not that good. I will change it a little bit and fix those button... Still I'm not sure where to put the shuffle drawing button that you are talking about. I will put it in the home page for the moment just to have it and test it. With the usage we will see where it is the best place to put it in.

> And concerning the save issue without undo is not a problem.. the worth thing can happen is the customer need to do again the playlist...so no a big problem 😉

I'm more and more convinced like this every day... I hope to put some of the changes in place in these days 😉

vincentbraillard, 2021-03-05:
Hello
Have done some test with the new version of save issues. it's seem to save automaticaliy now but it still ask me if I wan to save each time. please see my video bellow! is it normal?

https://youtu.be/Xdddn9ioVHg

texx00, 2021-03-05:
Hi, in the version I pushed yesterday that message has been removed completely... Are you sure you are running the right branch? Te branch name is "playlist-fixes"

The changes are not on master still... And I need to do some more changes to fix some issues with the autosaving (maybe I will do it later today) in the banch.

vincentbraillard, 2021-03-05:
Oh sorry. I update master 🤦‍♂️ so when push to.master I will test for you

texx00, 2021-03-06:
Hi

I've been working a little bit but still I'm not ready to merge. 
To recap, at the moment I fixed points number 2, and 8.
Number 3 and 6 do not need a fix.

Still need to fix 1, 4, 5, 7, 9, 10.

I do not know how long it will take but I will keep you updated.

vincentbraillard, 2021-03-06:
Hello

I have made some test for you of the playlists_fixes branch. I noticed followinf issues

A) when creating new playlist, I enter the name but no way so save it! If I press "enter" it make a carriage return. It would be nice by pressing enter, it save it. 
B) Maybe you could make more visible the place where to write the playlist name!
C) when adding drawing from draw and no playist is already created, you should check it to desable the add to selected playlist and insert a "no playlist ceated yet"

I'm not able to creat anymore a playlist. I'm doing something wrong or do you already know this error

vincentbraillard, 2021-03-07:
Hi
Just tested the be323ef. I cannot anymore connect me to the server. I's like the web page is reloading like a while(true)

texx00, 2021-03-07:
> Just tested the be323ef. I cannot anymore connect me to the server. I's like the web page is reloading like a while(true)

This is about  a change I added in the last commits. It requires to use the install script on every pull.
I'm thinking about this change because if it doesn't work correctly it ends up in a loop like you are facing. The problem I'm aiming to fix is to clear the browser cache automatically after an update. Indeed, without forcing the cache clearing mechanism, after an update the browser will still use old files creating problems.

Are you sure you run the install script? If yes I need to change this code otherwise I will add a troubleshooting entry for this (for the moment at least.... I prefer to focus on other things and optimize this later in this case)

About the other points: i should have fixed also number 4 and 5. I also fixed letter A and E (not able to create the playlist)

About A: I need to apply something similar also in the options (already on the list)
About B: I have been thinking about that since the beginning. I also think like this is not really clear but I don't like the idea of putting a text input for the task. Do you have any suggestion on how it could be easier to understand that the field can be changed?
About C: added to the TODO list
About D: Still there but I think how I can fix it. I needed to fix the other stuff first.

Still missing to fix: 1,7,9,10, B, C, D

Let me know about the loop issue. I will work on the other problems in the meanwhile

texx00, 2021-03-07:
Fixed 1, C and D

Still missing 7, 9, 10 and B

I think I will also improve the modal to choose the elements that one wants to add. At the moment with the icons only it's not easily understandable.

vincentbraillard, 2021-03-07:
Hello.
I test the last update. You did a incredible job incredible 👏
I just noticed small thing

Concerning point 1: if we want load a uploaded drawing, a black element bix appear with a waiting cercle. But if we click in it it open

texx00, 2021-03-09:
Sorry if I didn't answer before.

> Concerning point 1: if we want load a uploaded drawing, a black element bix appear with a waiting cercle. But if we click in it it open the popup windows!

> Then open the current drawing in single view doesn't work!

They both work on my computer (I tried on 2 different computers also)
Try with the new version

> Concerning timing element you may should directly open the setting popup. Now we need to creat the element and then ckick again in it to set a time! Why nit directly open the setting window directly? Same for random element or start playlist still concerning the timing element. When we set a time and press enter it exit without saving

I added the auto-open thing after creation. Still I do not understand why the timing element is not saved correctly. I need to dive into it more in depth.

The random element is substituting itself with a random drawing from the full list or from the entire playlist (so if you put in the middle it may look for the first half or the last half regardless). The thing is that is not shuffling the playlist but is putting in that position a drawing taken randomly among the others or from the complete list.

texx00, 2021-03-13:
I added this:
[image]

At the moment it works only for the drawings but I'm going to add it inside the playlist page as well.
Still I'm not completely sure about the design. There is something not convincing me completely

It fixes 9 and 10.
I still miss to fix the queue page (7) and the playlist name.

In the meanwhile, if you get a chance to test the changes let me know. I will merge the changes to get the playlists updated to the last version also in the master branch. I think I will still add the change for the playlist shuffle but the queue changes will go in another pull request

texx00, 2021-03-14:
I added the option to the playlists.

I added an option in settings to autostart automatically drawings when the device is turned on.

I think this are the last changes for this branch. I will only add fixes related to the changes now. About the queue, as I said, I will create a new branch because it is quite a big change.

Let me know when you test it so that I can merge with the master branch.

closed 2021-03-14 by texx00

vincentbraillard, 2021-03-24:
Hello

I Have found a bug with the playlist.
I have made a playlist with around 10 draw and at the end a playlist play action the the same playlist.
I have try shuffle play mode and normal playmode. It begin to start normally to draw but at the end, it doesn't follow with the next drawing.
At the bottom I have the message: Elements sarted, Elements ended showing while true...
I have try with a waiting time of 600s, after the first draw, on the now playing tab, I see the clock but nothing start after 600s
I have also try with 0s timing, after the first draw, I can also see on the now playing tab the clock and nothing start!

texx00, 2021-03-24:
I think I messed up a bit the things with the new play/shuffle controls... I need to rethink that, also because I'm not completely satisfied...

PR #35 (2021-03-14): Playlist fixes
Once ready, will close #34.

For the moment added only autosave capability