texx00/sandypi #40: Error on index for path drawing widget[BUG]

vincentbraillard, 2021-03-27:
**Describe the bug**
When I add new drawing, the widget are well. but when I delete one and just after I upload a new one, the visualisation doesn't fit with the file. Like a index shift!

**To Reproduce**
Steps to reproduce the behavior:
see upside

**Expected behavior**
be the right drawing shown

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - server OS: [Raspbian OS]
 - server hardware [ Raspberry Pi 3B+, ....]
 - serial device firmware [Grbl, ...] + version
 - Version [use `git rev-parse --short HEAD`]

**Additional context**
Could you use a thinner pen to draw the path because with complex path, the widget is full white

texx00, 2021-04-06:
I was not working on this, but I found out the origin of the problem. The problem is that when the last uploaded drawing is deleted, the same id is used again in the next uploaded drawing. This means that the frontend will use the old cached image instead of using the fresh one. I'm trying to make it work with the autoincrement of the drawing id but sqlite is not working as expecting (the autoincrement option for the primary key should fix the problem but I do not know why it is not working as expected). I will put in place a workaround (not the most elegant solution but the right way is not working, don't understand why. Should be fixed from commit ce34405. Will be available to test with the queue page update

vincentbraillard, 2021-04-06:
Hello

Unfortunately it didn't work. It's maybe worse. I had to upload 4 times to
have the right index (draw) see bellow. It is the last firmware ce34405
<https://github.com/texx00/sandypi/commit/ce34405a7937575bc92e2e89f50f7cd0a06350d1>

texx00, 2021-04-07:
I cannot see any picture, sorry.
To finally fix the problem you must clear the browser cache. If you do that it should work.
I got an idea that may fix also this thing about the browser cache not cleared between updates. I will try to put it in soon.

vincentbraillard, 2021-04-07:
Sorry. Here the picture...

vincentbraillard, 2021-04-07:
Hello

Clearing the cache solve the issue... ;-)

texx00, 2021-07-04:
Should be fixed

commit ce34405a (2021-04-06): Upgrading db for a better queue. Should also fix #40
closed 2021-07-04 by texx00