texx00/sandypi #43: Error to start the server or installation

vincentbraillard, 2021-06-22:
Necessary details:
* server OS (Raspbian OS, ...):
* server hardware (Raspberry pi 3B+, ...):
* serial device firmware (Marlin, Grbl) + version: 
* commit used (hash) (use the command `git rev-parse --short HEAD`):

Description of the problem:
Hello Luca
From 2 days now, I have a issue with sandypi. I Have done nothing but sudantily, the sandyfi don't start any more. I have try to start and also to reinstall the alpha branch without success.

I have made a txt file with a copy of the start and installation script.
Did you know why?

Necessary attachments:
[start_error.txt](https://github.com/texx00/sandypi/files/6696639/start_error.txt)

texx00, 2021-06-22:
Hi, are you running it from an old installation? With a first look it's something about the saved settings. Can you try to save the "saved_setting.json" file somewhere and delete it from the saves folder (it should be created again from scratch)? Still the same error? In that case, can you post here your saved_settings file?

vincentbraillard, 2021-06-22:
ok
I did the first solution and it work. but I have changed nothing in this file before and after the issue. but My settings file was......empty. that's crazy

Thanks Luca in any case

closed 2021-06-22 by vincentbraillard