texx00/sandypi #57: [BUG]

vincentbraillard, 2021-09-08:
**Describe the bug**
Fail to install firmware on raspberry pi 3b or 3b+ using node V 12.X.

**To Reproduce**
Steps to reproduce the behavior:
install node 12.x
Install sandypi

**Expected behavior**
fail to install. Fatal error
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - server hardware [Raspberry Pi 3B+,3B]

**Solution**
For raspberry Pi 3b or 3b+ you need to install node version 10.X following this link:
[https://noviello.it/come-installare-node-js-e-npm-su-raspberry-pi/](url)

texx00, 2021-09-09:
Glad you found a solution... When I get the time I will check it out better to see if there is a way to run it on v12 or v14...

texx00, 2022-01-06:
Closed by  #64

texx00, 2022-01-06:
Closed by #64

PR #64 (2022-01-06): Adding docker support
The pull request is to keep track of the "road to docker" changes.
At the moment the software runs in docker with a multistage image and running single installation commands instead of using the old install.sh script (different approach with respect to #36) .

Work to be done:
 - use volumes to keep drawings, database and conf files available even after an image update
 - use buildx to prepare raspi compatible images
 - create a docker-compose file to run the image with the correct ports and to add access to hw device
 - create docs

When this branch is finished, docker i [... cut]
closed 2022-01-06 by texx00