texx00/sandypi #62: LED Initialization after Drawing Start

tigert10, 2021-11-21:
**Describe the bug**
If LEDs are turned on after a drawing is started the system freezes and must be rebooted.

**To Reproduce**
Steps to reproduce the behavior:
1. Power on the system.
2. Turn LEDs on and everything works great, I can control color and brightness while the drawing is going with no problems.
However, if I:
1. Power on the system.
2. Start a drawing.
3. Then, try to turn LEDs on while drawing is running, the system freezes and must be rebooted.

**Expected behavior**
I expect to be able to start a drawing automatically on boot and turn the LEDs on without it freezing the system.

**Platform (please complete the following information):**
 - server OS: Raspian
 - server hardware Raspberry Pi 4
 - serial device firmware: Alpha 
 - Version: Alpha 12a9442

texx00, 2021-11-21:
I will have a look. Can you help me with the log files? There is a server/logs folder with some .log files... Just the feeder logs should be enough

tigert10, 2021-11-22:
You bet! Hope this is what you were looking for.
Let me know if there is anything else I can do. My coding experience is
pretty much limited to some webpage html and css and some arduino code, but
if I can help with anything let me know.
Thanks,
Ty

tigert10, 2021-12-08:
So this bug may only exist in the alpha version. I recently "downgraded" with a fresh install of the Master version and the problem does not seem to be appearing. Just FYI

texx00, 2021-12-08:
Thanks, good to know