texx00/sandypi #63: HELP! Server Won't Start

tigert10, 2021-11-24:
Necessary details:
Raspbian
RPI4

Description of the problem:

I have tried a fresh install of SandyPi and it is getting hung up at the same spot. Seems to be a serial port problem but I'm at a loss as to how to fix. The install instructions talk about disregarding serial port warnings until you set the serial port in the GUI, but I can't get to the GUI as it won't initialize. It initializes until it gets to (maybe typos here as I typed it instead of copy/paste):

"ERROR in server.hw_controller.feeder (device_serial.py): Port must be configured before it can be used.
Traceback (most recent call last):
File "/home/pi/sandypi/server/hw_controller/device_serial.py", line 34, in _init_
self.serial.open()
File "/usr/local/lib/python3.9/dist-packages/serial/serialposix.py", line 316, in open
raise SerialException("Port must be configured before it can be used.")
ERROR in server.hw_controller.feeder (device_serial.py): Serial not available. Ar you sure the device is connected and is not in use by other softwares? (Will use the fake serial)
/home/pi/sandypi/server/hw_controller/leds/leds_types/dimmable.py:29; RuntimeWarning: This channel is already in use, continuing anyway. Use GPIO.setwarnins(False) to disable warnings.
GPIO.setup(self.pin, GPIO.OUT)
INFO:server:Leds controller started

texx00, 2021-11-24:
Hi! You can try two things:
- did you set the autostart to be on? try to reboot the pi and use the "top" command to check if you have a flask service running already... If the service is running you should be able to connect directly without the need of starting it again even after a new installation
- try to delete (if you want make a backup copy before deleting it) the saved_settings.json file under server/saves (do not delete the default_settings.json) and run again the software

tigert10, 2021-11-24:
Hi, tried both of your suggested solutions with no success.
I have identified that the problem is in the Frontend part of the install. 
Under the "Installing js dependencies and building frontend app" there seem to be many errors. But not sure how to solve them. I have included the install log file. Hope it helps. I look forward to your reply!
[install.log](https://github.com/texx00/sandypi/files/7597106/install.log)

tigert10, 2021-11-24:
So in following your advice to another person with a similar issue here:
https://github.com/texx00/sandypi/issues/26
I tried doing the frontend yarn build trick and it gave an error. I have attached a txt file with the error. Hope it helps.
Thanks in advance for all your help!
[error.txt](https://github.com/texx00/sandypi/files/7597708/error.txt)

texx00, 2021-11-24:
Looks like you are missing node-sass which is a library required during the installation. 
Not sure the reason why is not installed... Try to have a look at this if it works to fix the node-sass library. Otherwise you may need to build it or google some more about the installation on your version...
Hopefully problems like this will be fixed with #8 which is one of the priorities at the moment... Just need the time to set it up

tigert10, 2021-11-24:
Got it! You were absolutely correct about the node-sass being the culprit. I think the problem stems from the node version that I was using. I could be wrong but I was using Node 17 and I don't think node-sass was compatible. After much troubleshooting I downgraded node to 12 and reinstalled SandyPi and behold it works. Thanks for your prompt replies and helpful responses. I really appreciate your work on this project! It is awesome!

texx00, 2021-11-24:
Thanks! Since it is solved I will close it

closed 2021-11-24 by texx00