texx00/sandypi #93: Error during docker installation : ImportError: cannot import name 'ContextAPI' from 'docker'

ydakilux, 2024-01-31:
Necessary details:
* server OS (Windows, Raspbian OS, ...): 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
 Kernel: Linux 6.1.21-v7+

* server hardware (PC, Raspberry pi 3B+, ...): **Raspberry pi 3B**
* serial device firmware (Marlin, Grbl) + version: **Marlin**

Description of the problem: 
Clean install from Raspberry OS & docker.
I'm unable to instal SandyPi with new installation method from https://github.com/texx00/sandypi/blob/master/docker/readme.md

Docker and docker compose are installed

**Docker version 20.10.5+dfsg1, build 55c4c88**
**Docker Compose version v2.24.2**

Necessary attachments:

Log history :
docker-compose up -d
Traceback (most recent call last):
  File "/usr/local/bin/docker-compose", line 5, in <module>
    from compose.cli.main import main
  File "/usr/local/lib/python3.9/dist-packages/compose/cli/main.py", line 29, in <module>
    from ..metrics.decorator import metrics
  File "/usr/local/lib/python3.9/dist-packages/compose/metrics/decorator.py", line 3, in <module>
    from compose.metrics.client import MetricsCommand
  File "/usr/local/lib/python3.9/dist-packages/compose/metrics/client.py", line 5, in <module>
    from docker import ContextAPI
ImportError: cannot import name 'ContextAPI' from 'docker' (/usr/local/lib/python3.9/dist-packages/docker/__init__.py)

Help welcome, I'm a newbies with docker

texx00, 2024-01-31:
Hi!

The only thing I was able to find about this problem is this issue:
https://github.com/docker/compose/issues/8366

Are you sure docker-compose and docker are installed correctly?
You may try to run some other image as a test, just to understand if the problem is with your setup or this image...

ydakilux, 2024-02-03:
Hello,
 
  I tried the classic hello world docker and it worked fine.
  
  I've tried and succeeded to launch a docker and a docker-compose from scractch using this method ( https://docs.docker.com/compose/gettingstarted/#step-1-define-the-application-dependencies)
 
 I therefore assume that the docker and docker-compose are correctly installed. But don't know what's wrong
  
 Output of docker version:
```
 Client:
 Version:           20.10.5+dfsg1
 API version:       1.41
 Go version:        go1.15.15
 Git commit:        55c4c88
 Built:             Mon May 30 18:34:49 2022
 OS/Arch:           linux/arm
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.5+dfsg1
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.15.15
  Git commit:       363e9a8
  Built:            Mon May 30 18:34:49 2022
  OS/Arch:          linux/arm
  Experimental:     false
 containerd:
  Version:          1.4.13~ds1
  GitCommit:        1.4.13~ds1-1~deb11u4
 runc:
  Version:          1.0.0~rc93+ds1
  GitCommit:        1.0.0~rc93+ds1-5+deb11u2
 docker-init:
  Version:          0.19.0
  GitCommit:
```

 **Output of docker-compose config**
```
 Traceback (most recent call last):
  File "/usr/local/bin/docker-compose", line 5, in <module>
    from compose.cli.main import main
  File "/usr/local/lib/python3.9/dist-packages/compose/cli/main.py", line 29, in <module>
    from ..metrics.decorator import metrics
  File "/usr/local/lib/python3.9/dist-packages/compose/metrics/decorator.py", line 3, in <module>
    from compose.metrics.client import MetricsCommand
  File "/usr/local/lib/python3.9/dist-packages/compose/metrics/client.py", line 5, in <module>
    from docker import ContextAPI
ImportError: cannot import name 'ContextAPI' from 'docker' (/usr/local/lib/python3.9/dist-packages/docker/__init__.py)

```

texx00, 2024-02-03:
Ok, I think now I understand better...

The compose module used to be a separate installation from docker and it was run with docker-compose (notice the dash). The new version of docker comes with the compose module already installed/integrate: you should try using docker compose (without dash).
The command should work the same but I never tried. It is should be still possible to install the docker-compose module manually (I didn't see it mentioned in the link you posted, I assume you don't have it), but I would try the other way first.
If this doesn't work, please paste herethe result of running "docker-compose --version" (once with and once without dash)

ydakilux, 2024-02-03:
In between, I tested the following:

```
pip uninstall docker 
pip install docker==6.1.3
```

And now SandPi is running :-)
I will try to go back to docker 7.0.0 and will keep you posted with the docker compose (without the dash)

ydakilux, 2024-02-04:
Last update.

I made a fresh installation with the Raspberry Pi Image (v1.8.5), taken the raspebrry Pi OS (Legacy, 32 bit).
Followed this guide to instal docker (https://docs.docker.com/engine/install/raspberry-pi-os/)

Run those 3 lines (no sure if they are completly necessary)

```
sudo groupadd docker
sudo usermod -aG docker ${USER}
su -s ${USER}
```

Run also this line to run fix permissions problem:

`sudo chmod 666 /var/run/docker.sock`

And finally run the following command (without the dash in docker-compose)

`docker compose up -d`

SandyPi is now running

Now I need to connect to my marlin board :-)

texx00, 2024-02-04:
Perfect! Thank you for the update

closed 2024-02-04 by texx00